mjrice/Wilson2 #5: Missing extruder fan duct

WardBenjamin, 2015-09-10:
When assembling a list of parts (by filename) to print based on the BOM, I found that you are missing the stl for the fan duct. I know there is one because there's one in the BOM as well as the picture on reddit, but apparently it didn't get added.

WardBenjamin, 2015-09-13:
I saw that you have added a fan duct specifically for the e3dlite6 hot end. Will this work for other hot ends, such as a hexagon or a jhead?

mjrice, 2015-12-12:
For the head hot end you would probably want to use the other fan duct which is located in the original Wilson repo in the stl/extruder folder. I am not sure which would work for a hexagon, it depends on the length.