mjrice/Wilson2 #12: [Question] SCAD file of every part?

andreq, 2015-10-17:
I'm trying to make a Wilson 1.5 (mixing TS and 2 parts), but having a hard time using sketch up.
I do most of my design on Solidworks and OpenSCAD. 

Is there plan to have SCAD file for all the parts?

mjrice, 2015-10-22:
I don't have any plans to re-design parts like the Y ends in scad.

andreq, 2015-10-22:
Thanks for the info. 

I'm currently starting from scratch in Solidworks, I'll try to branch out from Wilson TS or host the file somewhere if I get anywhere with it. 

I'm only looking to enhance the Z frame with the 2nd set of extrusion. It look way more solid on the Wilson 2 :).

Btw, thanks for designing those nice printer, I'm happy I chose the Wilson TS back in April!

closed 2015-10-22 by andreq