mjrice/Wilson2 #15: Y ends have oversize holes for 10mm rods

hasbridge, 2015-11-25:
During assembly, I noticed that my 10mm rods are loose in the Y end pieces. I imported both STL files into sketchup, and each hole is oversized. On the y-motor stl, they appear to be 10.5mm, and on the y-idler piece, they appear to be 10.8.

thatch, 2015-12-26:
They are indeed oversized on the one I printed as well.  I'm switching to a more Taz-style Y end, as this piece is the toughest to print.

mjrice, 2015-12-26:
The idler end holes are indeed a little over sized, and I'll fix that tonight and push an update.  The .5mm on the motor end holes was intentional - with a 0.5mm nozzle, interior holes tend to be undersized by the diameter of the nozzle.  Maybe I should generate a couple different variations for smaller nozzles.

hasbridge, 2015-12-29:
I would actually prefer undersized by a tiny bit, because I could ream them with a 10mm drill bit to get a perfect 10mm hole afterwards.