mjrice/Wilson2 #18: Issue slicing new z-lower-brace-1 and z-lower-brace-2

tamarintech, 2015-12-15:
When using Cura 15.02 to print the new lower braces they do not slice correctly.

z-lower-brace-1:
[image]

z-lower-brace-2 doesn't seem to load at all.

z-lower-brace-pair loads correctly and slices as expected.

I'm not sure if this is a Cura issue or an issue with the Sketchup Exporter.  I was able to print the braces after re-exporting them with Blender.  If other users experience the same issue I will gladly open a PR for the reexported braces.

Thanks!

thatch, 2015-12-26:
I have the same issue with Cura 15.04, including on z-brace-lower-pair.  I think it's mostly cosmetic, but it would be good to have a fixed version in the repo (either blender or netfabb exported).  (A friend doesn't see the error on his using s3d.)

thatch, 2015-12-26:
I checked and netfabb does not repair this issue, and neither do any of the "Fix horrible" expert options in Cura 15.04.  Can you open the PR?

Arunk3001, 2015-12-26:
I believe it's a issue with your settings in Cura. If you open the expert config window (MenuBar>Expert>open expert settings...) you should find in the lower right corner a section titled "Fix horrible". If I'm right, you have Combine everything(Type-B) selected. This setting fills in any gap between model walls, including space meant to be hollow. Try unchecking that setting and try slicing it again.

tamarintech, 2015-12-26:
Sorry, @Arunk3001 but please check the github viewer.  You'll find one of the braces actually has all of the faces flipped.

It looks like Marty pushed the braces through netfabb not long ago.  If that doesn't work for you then these are what I used:
https://github.com/tamarintech/Wilson2/blob/lower-braces/stl/z-lower-brace-1-reexport.stl
https://github.com/tamarintech/Wilson2/blob/lower-braces/stl/z-lower-brace-2-reexport.stl

Here's the most recent from Marty https://github.com/mjrice/Wilson2/commit/54ae108adc2ba3e7dde95da91132310c9987c57a

tamarintech, 2016-01-06:
I believe this is solved.
Closing.

closed 2016-01-06 by tamarintech