mjrice/Wilson2 #21: Lead screw length is not a formula in the BOM

tamarintech, 2015-12-22:
I built to 400mm and seem to be needing 385mm lead screws.

Is Z height - 15mm a sane value for NEMA 17 with a coupler?

hasbridge, 2015-12-22:
My leadscrews (300mm) are about 20mm off the top of the motor mount - they don't need to go all the way down.

tamarintech, 2015-12-22:
You mean all the way down in the z-coupler?  Yea, I shoved mine all the way down.  I'd still end up with a pretty big gap.

What should the formula be?  Z - 15?  Z - 20?

hasbridge, 2015-12-22:
I am confused - are you saying your vertical extrusions are 400mm tall, or 400mm build height? For 400mm long extrusions (what I used), then 300 is the correct length for the lead screws. I would think the formula would always be Z - 100, with Z representing the total vertical extrusion length.

tamarintech, 2015-12-22:
Z rod: 400mm
Z axis extrusions (vertical frame piece in sheet): 480mm
Z Print height: 280mm

My lead screws are 350mm and 35mm away from the bearing trap - so that would mean I need 385mm lead screws.  480mm - 100mm sounds good - that would put me at 380mm which should fit in the bearing trap.

Should we get a note or a formula cell added to the sheet for Z axis extrusion - 100 = lead screw length?

Cell C19 would be =(C11-100) ?

hasbridge, 2015-12-22:
Yeah, I think that should work out correct. There should be a slight gap between the leadscrew and motor shaft inside the coupler, so that would account for the extra 5mm.

tamarintech, 2015-12-22:
https://github.com/mjrice/Wilson2/pull/23

E19 (lead screw length) updated to =C$11-(IF(B$3="metric",100,4)) where C11 is "Vertical frame piece" extrusion.

tamarintech, 2015-12-22:
Marty did it a bit more gracefully. ;)

https://github.com/mjrice/Wilson2/commit/cf690a2f3e09eeb61adfe3a06d8e54011543dffe

closed 2015-12-22 by tamarintech