mjrice/Wilson2 #24: Z rods ~5mm too short

thatch, 2015-12-26:
Using the lengths from the BOM, and putting the z-motor parts against the table, the rods are about 5mm too short.  I can compensate for this by cutting them longer, it just seems like a doc problem.  Originally noticed by @precision with s3d and confirmed with my re-print using Cura.

mjrice, 2015-12-26:
The pocket on the top Z rod holders should have some room to spare with the 320mm smooth rod.  Either a few mm shorter or longer is meant to work.  Do you rods actually not reach the pocket?

precision, 2015-12-26:
Yeah, my right Z rod in particular was short and just barely reached the pocket.  I ended up sticking a M5 nut on top of the rod to snug up the fit.  Without the nut, it would wiggle itself out of the pocket. This is on Misumi cut extrusion and rods, btw.

mjrice, 2015-12-26:
Thanks for the feedback - I modified the z tops to extend the rod holder down another 5mm.

thatch, 2015-12-27:
Is your intention to make it so you can slip the rod into the top pocket while the printed parts are already on the extrusion?  If not, I think the top pocket needs to be at least 2mm shallower (and you can make the bottom one deeper by the same amount if you're concerned about fitting existing cut-too-long rods).

I measured mine today (not including your latest modification), and surface-to-surface is 310mm, bottom pocket is 4.7mm, top pocket is 11mm.  So with the rod pushed fully up, I have 1mm gap at the bottom.

thatch, 2017-10-18:
I'm no longer building a Wilson2, so closing.

closed 2017-10-18 by thatch